JacekDob/Sonoff-Tasmota #1: Compilation error

ionciubotaru, 2018-02-18:
In webserver.ino `func += String(ESP.getChipId())` failed to compile. I comment the line and everything works fine. I use the latest esp32 core from https://github.com/espressif/arduino-esp32

Please add `ESP32Ticker` and `WebServer-esp32` in lib folder

It will be great if you can update your fork to main Tasmota's updates.

JacekDob, 2018-02-22:
A little bit strange as in sonoff.ino there is:
```
#define getChipId() getEfuseMac()
```
as I didn't find get chip id for ESP32.
If it works in sonoff.ino (where is used couple of times, should also work in webserver.ino).
I am using Arduino 1.8.5.

Libraries added.

Updated to v5.12.0c. There are so often changes in main Tasmota, so keeping very up-to-date is hard. Contribution welcome.

ionciubotaru, 2018-02-22:
Great update, It's not important to update your fork on every small release, only on major releases.
Another small suggestion, add a link to your  ESP32 fork in Tasmota's wiki - a new page on the left side.
In this way more people will notice about your fork. The more the community the more contributors.

JacekDob, 2018-02-22:
Added on main page on right side topic list.

flexiti, 2018-02-25:
Firstly, I have such an error:
[image]
second, All the time I see that everything compiles on ESP8266, does something else have to be set to compile in the ESP32 environment? (my board is ESP32 devkit v1 from doit)

JacekDob, 2018-02-25:
Did you change board to ESP32 dev module?

JacekDob, 2018-02-25:
An what version of IDE do you have? I am using 1.8.5.

flexiti, 2018-02-25:
@JacekDob 
Thanks for help

I thought that from default is ESP32 :)

You mean I have to change  board = esp01_1m to eg. my board board = esp32doit-devkit-v1 ??

eg.
_[env:sonoff]
platform = espressif8266
framework = arduino
**board = esp32doit-devkit-v1**
board_flash_mode = dout
build_flags = -Wl,-Tesp8266.flash.1m0.ld
lib_deps = PubSubClient, NeoPixelBus, IRremoteESP8266, ArduinoJSON
extra_scripts = pio/strip-floats.py_

I am using arduino 1.8.5
I'm a bit confused because I can not find any ESP32 settings in the project, only _ifdef ESP32_

Mayby small how to -  my board ESP32 devkit v1 from doit ? :)

JacekDob, 2018-02-25:
Install ESP32 for Arduino:
https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md

In Arduino IDE set Tool->Board->ESP32 Dev Module
[image]

Add in user_config_override.h:
```
#undef USE_ADC_VCC    // Needs to be ported
#undef USE_IR_REMOTE  // IRremoteESP8266 needs to be ported
#undef SNFBRIDGE      // No commercial device, no sense porting
```

Copy all libraries in \lib to Arduino\libraries

Add support of unsigned long long in WString as in Readme.

JacekDob, 2018-02-25:
platformio.ini is not for Arduino IDE and is not used at all

JacekDob, 2018-02-25:
I have updates change is user_config_override.h so please refresh.

I have also added installation instructions in project Readme.

flexiti, 2018-02-25:
Thanks, it works a little  ;)
I thought it was prepared for the Platformo environment, hence the misunderstanding.
its compiles under Arduino IDE.

For now I have noticed that it do not want to work on the static address eg. 192.168.0.39
, with DHCP working. (WIFI_IP_ADDRESS) I put it off today and I will make sure tomorrow.
Thanks for help and for the port on ESP32 (I use a lot of ESP8266, this is the first attempt on ESP32)

:)

JacekDob, 2018-02-26:
Porting to work with Platformo to be done.

Static should work in terms of implementation, support.ino:
```
 if (Settings.ip_address[0]) {
    WiFi.config(Settings.ip_address[0], Settings.ip_address[1], Settings.ip_address[2], Settings.ip_address[3]);  // Set static IP
  }
```

flexiti, 2018-02-26:
if WIFI_IP_ADDRESS = "0.0.0.0" works,
if WIFI_IP_ADDRESS = "192.168.0.39" not (see below)
I checked, this address is not occupied and work if compiled on ESP8266
It looks like there is a problem logging into the wifi if a static address is set
Can you check if your ESP behaves differently?

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58

00:00:00 APP: Set Baudrate to 115200

00:00:00 Project sonoff Test (Topic sonoff, Fallback Test, GroupTopic sonoffs) Version 5.12.0c-STAGED
00:00:00 WIF: Connecting to AP1 T100 in mode 11
00:00:15 WIF: Connect failed with AP timeout
00:00:15 WIF: Connecting to AP1 T100 in mode 11
00:00:18 WIF: Connect failed as AP cannot be reached
00:00:19 WIF: Connect failed as AP cannot be reached
00:00:20 WIF: Connect failed as AP cannot be reached
00:00:21 WIF: Connect failed as AP cannot be reached
00:00:22 WIF: Connect failed as AP cannot be reached
00:00:23 WIF: Connect failed as AP cannot be reached
```

JacekDob, 2018-02-26:
Are you setting also WIFI_GATEWAY, WIFI_SUBNETMASK, WIFI_DNS with proper values?

Zixim, 2018-02-26:
I'm having trouble compiling, these are just some of the errors I get.
```

C:\Users\Tom\Documents\Arduino\hardware\espressif\esp32\cores\esp32/WString.h:58:5: note: candidate: String::String(const String&)

     String(const String &str);

     ^

C:\Users\Tom\AppData\Local\Temp\arduino_modified_sketch_32088\xplg_wemohue.ino: In function 'void HueLights(String*)':

C:\Users\Tom\AppData\Local\Temp\arduino_modified_sketch_32088\xplg_wemohue.ino:698:18: warning: switch condition has type bool [-Wswitch-bool]

         switch(on)

                  ^

Multiple libraries were found for "WiFi.h"
 Used: C:\Users\Tom\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi
 Not used: C:\Users\Tom\Downloads\arduino-1.8.4\libraries\WiFi
exit status 1
call of overloaded 'String(uint64_t)' is ambiguous

```

Could you please please consider uploading the .bin files required for using this ?

flexiti, 2018-02-26:
@JacekDob 
Yes, WIFI_GATEWAY, WIFI_SUBNETMASK, WIFI_DNS are set corectly
eg. in user_config_override.h
```
#ifdef WIFI_GATEWAY
#undef WIFI_GATEWAY
#endif
#define WIFI_GATEWAY           "192.168.0.1"     // {IpAddress2] If not using DHCP set Gateway IP address

#ifdef WIFI_SUBNETMASK
#undef WIFI_SUBNETMASK
#endif
#define WIFI_SUBNETMASK        "255.255.255.0"   // [IpAddress3] If not using DHCP set Network mask

#ifdef WIFI_DNS
#undef WIFI_DNS
#endif
#define WIFI_DNS               "8.8.8.8"

```
Static address work ok. on Your's ESP ?

JacekDob, 2018-02-28:
Then it needs analysis.

Try simple example for ESP32: 
[image]

And check if it works standalone.

flexiti, 2018-02-28:
The same effect,
if IP 0.0.0.0 works, if 192.168.0.38 is not.
Does this example work for you?

JacekDob, 2018-03-01:
I haven’t tried it. I’m using dhcp with static lease on router. I will try to test it. Look on arduino esp32 github if it was reported.

flexiti, 2018-03-01:
Even if you use DHCP, you can use a static address, just make sure it is not in the DHCP pool. I would like to confirm only if it is not my hardware fault. You can check it ?

flexiti, 2018-03-01:
Static IP
This:
_https://github.com/espressif/esp-idf/pull/657/files#diff-6a4183dba634c596fdfa03b23c74a8d2L746_

fix this error, now is ok

JacekDob, 2018-03-05:
Should issue be closed?

flexiti, 2018-03-06:
Yes, so far all is working ;)

closed 2018-03-06 by JacekDob