JacekDob/Sonoff-Tasmota #3: compiled .bin images

Zixim, 2018-02-19:
Could some one upload compiled .bin images, just like in the Original Tasmota fork ?

JacekDob, 2018-02-22:
One image released:
https://github.com/JacekDob/Sonoff-Tasmota/releases/tag/v5.12.0

Zixim, 2018-02-25:
The esp32 I have is brand new out of box, well...plastic baggie 😛 
Never been flashed by me...
Can I flash the firmware with settings as per screenshot ?
Or do I need to worry about bootloader & partitions & other bins ?

[image]

JacekDob, 2018-02-25:
Not sure. Always I've been using Arduino IDE and uploading does everything automatically. But rather bootloader + partitions + boot_app0 + app. I may try to upload them tommorow.

Bootloader and boot_app0 is in Arduino SDK. Partitions and app come from compilation. I would say it is quicker to download Arduino IDE, copy libraries and upload from IDE.

JacekDob, 2018-02-28:
Add all files in:
https://github.com/JacekDob/Sonoff-Tasmota/releases/tag/v5.12.0

JacekDob, 2018-03-02:
Does it work?

Zixim, 2018-03-03:
This works : 
[image]

JacekDob, 2018-03-05:
Should issue be closed?

closed 2018-03-05 by Zixim