JacekDob/Sonoff-Tasmota #4: wifi AP defaults should not be empty / why are they changed from Tasmota defaults

Zixim, 2018-03-02:
https://github.com/JacekDob/Sonoff-Tasmota/blob/ccc14966d50f583c96862c9fccc5dbd8e8d36c66/sonoff/user_config.h#L56-L59

This renders precompiled flashable .bin files useless, because there is no way to reach the flashed device.

JacekDob, 2018-03-03:
If you flashing via serial you can set all via serial.

Zixim, 2018-03-03:
Of course not my own ssid, as you well realise.
The default known ssid's & passwords that are in the original Tasmota are just fine, no?

I tried settting those parameters via serial, but that does not seem to work.

JacekDob, 2018-03-03:
What are Tasmota default ?

Zixim, 2018-03-03:
```
#define STA_SSID1              "indebuurt1"      // [Ssid1] Wifi SSID
#define STA_PASS1              "VnsqrtnrsddbrN"  // [Password1] Wifi password
#define STA_SSID2              "indebuurt2"      // [Ssid2] Optional alternate AP Wifi SSID
#define STA_PASS2              "VnsqrtnrsddbrN"  // [Password2] Optional alternate AP Wifi password
```
did this change recently perhaps ?

JacekDob, 2018-03-03:
From what I see in current development branch in user config and in override those options are empty.

Zixim, 2018-03-03:
I just checked this, you are absolutely right.
I used to flash the minimal tasmota bin, which has those default AP's preconfigured.
Then i OTA from minimal to current.

But failing this, I'm stuck on accessing the Tasmota32 for changing the AP settings...
Serial won't respond when not connected to wifi, and it won't connect to wifi because the parameters aren't set 🙄 

Any idea why serial isn't working ?

Zixim, 2018-03-03:
https://github.com/arendst/Sonoff-Tasmota/issues/2072

in short : Putty is useless, use Termite.
Which then allowed me to set the SSid and Password through a serial connection.

closed 2018-03-03 by Zixim