JacekDob/Sonoff-Tasmota #8: [FR] Soldering / wiring guide to upgrade sonoff

aenertia, 2018-03-10:
Am interested in seeing a pin/wiring guide to replace the esp 8266 on sonoff (basic/th) with an esp32. 

I have a few projects which would benefit from hardware pwm and extra pins and being able to solder into existing sonoff after removing the smd esp 8266 would be a great way to encourage interest in this port. 

Include appropriate esp dev board models etc.

flexiti, 2018-03-11:
I do not know exactly what you're asking, but maybe start from that
http://randomnerdtutorials.com/getting-started-with-esp32/

aenertia, 2018-03-11:
Looking for a Step by Step to replace the esp8266ex chip in place with an
esp32 in the sonoff.

If this were an accessible chunk of howto in one place for hacking up
existing sonoff pcb's (rather than a DIY with a Wemos/Esp32 Dev board +
relay etc) it would likely be well received.

Zixim, 2018-03-12:
you would need to remove the esp8266 from the pcb, then put in place some kind of break-out from the pads on pcb, then wire up the esp32 to that break-out.
You would be much better off making a homebrew device based on a dev-board esp32.

At least that's been done before and there's no micro-soldering involved.