JacekDob/Sonoff-Tasmota #9: ESP8266 user_config_overide.h

flexiti, 2018-03-11:
The second part was important to compile the same code for ESP8266.
Did you fix it elsewhere?

#ifdef ESP32
#undef USE_ADC_VCC    // Needs to be ported
#undef USE_IR_REMOTE  // IRremoteESP8266 needs to be ported
#undef SNFBRIDGE      // No commercial device, no sense porting
**#else
#define SNFBRIDGE
#define USE_ENERGY_SENSOR
#endif**

The second part was important to compile for ESP8266.
Did you fix it elsewhere?

JacekDob, 2018-03-18:
I have modified user_config_overide.h as you proposed.

I am not using this fork for ESP8266, so I am not testing it.

PS.
Skąd jesteś?

closed 2018-03-18 by JacekDob

flexiti, 2018-03-18:
Gdynia, dobrze że te źródła działają z ESP32, ale jak będą działać z ESP8266 jeszcze lepiej ;)   ( używam i ESP32 i ESP8266) .Dzięki za ESP32