JacekDob/Sonoff-Tasmota #12: PlatformIO config example

brandond, 2018-05-29:
What settings should I use in platformio.ini for esp32? I have a few different models, most of them Lolin32 knockoffs with varying pinouts but they all have the WROOM-32 version of the module.

brandond, 2018-05-29:
I tried the following:
```ini
[env:sonoff-esp32]
platform = espressif32
framework = arduino
board = lolin32
board_flash_mode = dio
lib_deps = PubSubClient, ArduinoJSON
extra_scripts = pio/strip-floats.py
```

Compilation fails with:
```
/mnt/media/repos/Sonoff-Tasmota-esp32/sonoff/sonoff.ino:343:28: error: variable or field 'WifiWpsStatusCallback' declared void
/mnt/media/repos/Sonoff-Tasmota-esp32/sonoff/sonoff.ino:343:28: error: 'wps_cb_status' was not declared in this scope
/mnt/media/repos/Sonoff-Tasmota-esp32/sonoff/webserver.ino: In function 'void HandleInformation()':
/mnt/media/repos/Sonoff-Tasmota-esp32/sonoff/webserver.ino:1934:69: error: call of overloaded 'String(uint64_t)' is ambiguous
func += F("}1" D_ESP_CHIP_ID "}2"); func += String(ESP.getChipId());
```

alexswi, 2018-07-10:
I had the same problem.

brandond, 2018-07-10:
I gave up and started using https://github.com/OttoWinter/esphomeyaml/